sifiveinc/meta-sifive #1: Is this SDK also appropriate for VC707 FPGA Dev Kit?

wwMark, 2019-10-10:
Hello, I would like to know whether this SDK is also appropriate for VC707 FPGA Dev Kit? Could I write the disk image to SD card with the zcat command as mentioned in the readme file or should I write the image in a different way if the SD card will be used in the VC707 FPGA Dev Kit? And how?

jim-wilson, 2019-10-14:
The vc707 FPGA dev kit does not support partitions on the SDcard.  OpenEmbedded requires multiple partitions for the kernel, opensbi, rootfs, etc.  So, no, I would not expect this to work.  As far as I know, the only thing that can work on the vc707 is the linker kernel with a buildroot initramfs linked into bbl as a payload, as this can work without partitions.  You can build one of these if you use the v1_0 branch of freedom-u-sdk.  After you build the image, you just dd bbl to the sdcard (not a partition but the whole sdcard), and then you should be able to boot it.

palmer-dabbelt, 2019-10-15:
We don't support the VC707 and don't have any plans to do so in the future.  Doing so wouldn't be that hard and I'd be happy to take a PR.

davidlt, 2020-03-31:
I am closing this issue as there is no plan to support VC707.

closed 2020-03-31 by davidlt