sifiveinc/meta-sifive #2: MACHINE=freedom-u540 is invalid

pmatos, 2020-01-14:
On Ubuntu 18.04, while trying to generate a bootable image for the Unleashed SD card, I get the following after the steps listed in the `README.md`:

```
➜  riscv-sifive $ . ./meta-sifive/setup.sh    
Init OE

### Shell environment set up for builds. ###

You can now run 'bitbake <target>'

Common targets are:
    core-image-minimal
    core-image-sato
    meta-toolchain
    meta-ide-support

You can also run generated qemu images with a command like 'runqemu qemux86'
Your build directory already has local configuration file!
If you want to start from scratch remove old build directory:

    rm -rf /home/pmatos/dev/riscv-sifive/build

---------------------------------------------------
MACHINE=freedom-u540 bitbake demo-coreip-cli
---------------------------------------------------

Buildable machine info
---------------------------------------------------
* freedom-u540: The SiFive HiFive Unleased board
* qemuriscv64: The 64-bit RISC-V machine
---------------------------------------------------
➜  build $ MACHINE=freedom-u540 bitbake demo-coreip-cli
ERROR:  OE-core's config sanity checker detected a potential misconfiguration.
    Either fix the cause of this error or at your own risk disable the checker (see sanity.conf).
    Following is the list of potential problems / advisories:

    MACHINE=freedom-u540 is invalid. Please set a valid MACHINE in your local.conf, environment or other configuration file.

Summary: There was 1 ERROR message shown, returning a non-zero exit code.
```

davidlt, 2020-01-14:
I launched `podman` rootless container on my Fedora 31 system with `ubuntu:18.04` (from Docker Hub). After solving problems in container (setting a proper locale and creating non-root user) it worked. I installed packages mentioned from https://www.yoctoproject.org/docs/2.7.1/ref-manual/ref-manual.html#ubuntu-packages and also `curl` (to download `repo` tool). I also installed `neovim` for editor.
```
[..]
NOTE: Starting bitbake server...
NOTE: Starting bitbake server...
Creating auto.conf
---------------------------------------------------
MACHINE=freedom-u540 bitbake demo-coreip-cli
---------------------------------------------------

Buildable machine info
---------------------------------------------------
* freedom-u540: The SiFive HiFive Unleased board
* qemuriscv64: The 64-bit RISC-V machine
---------------------------------------------------
testing@37c054be3980:~/riscv-sifive/build$ MACHINE=freedom-u540 bitbake demo-coreip-cli
Parsing recipes: 100% |##############################################################################################################################################################################################| Time: 0:01:37
Parsing of 2389 .bb files complete (0 cached, 2389 parsed). 3490 targets, 134 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.43.1"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-18.04"
TARGET_SYS           = "riscv64-oe-linux"
MACHINE              = "freedom-u540"
DISTRO               = "nodistro"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "riscv64 littleendian"
meta                 = "HEAD:6b36db836547a23f43c5f97bf3706d7b210c209c"
meta-oe              
meta-python          
meta-multimedia      
meta-networking      
meta-gnome           
meta-xfce            = "HEAD:4e0538516b1e0ef42dc79bd08f7895f0052063ac"
meta-riscv           = "HEAD:99cdf8d0cfe4b515ccac76c816091b146d0a012b"
meta-sifive          = "HEAD:87a209be777318fefb87da6e295f4c34540717f3"

NOTE: Fetching uninative binary shim from http://downloads.yoctoproject.org/releases/uninative/2.6/x86_64-nativesdk-libc.tar.xz;sha256sum=133387753a9acf3e1b788103c59fac91e968e2ee331d7a4b9498e926ada7be57
Initialising tasks: 100% |###########################################################################################################################################################################################| Time: 0:00:06
Sstate summary: Wanted 1950 Found 0 Missed 1950 Current 0 (0% match, 0% complete)
NOTE: Executing Tasks
[..]

testing@37c054be3980:~/riscv-sifive/build$ cat /etc/os-release 
NAME="Ubuntu"
VERSION="18.04.3 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.3 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic
```

Could you give another try? Especially on a clean environment.

jim-wilson, 2020-01-14:
In openembedded-core/meta/classes/sanity.bbclass I see
    # Check that the MACHINE is valid, if it is set
    machinevalid = True
    if d.getVar('MACHINE'):
        if not check_conf_exists("conf/machine/${MACHINE}.conf", d):
            status.addresult('MACHINE=%s is invalid. Please set a valid MACHINE in your local.conf, environment or other configuration file.\n' % (d.getVar('MACHINE')))
So it looks like the only way this can happen is if meta-sifive/conf/machine/freedom-u540.conf is missing.  Maybe check your repo tree to make sure it is OK?

pmatos, 2020-01-16:
Thanks @davidlt and @jim-wilson 
I might have missed some step setting this up. All working now.

closed 2020-01-16 by pmatos