sifiveinc/meta-sifive #3: Rebooting fails on Hifive Unleashed

thechristschn, 2020-01-30:
Hello,

I have a problem with rebooting my SiFive Unleashed.

I flashed my SD-card according to the instructions on sifive/freedom-u-sdk v201908-branch (which should be the same as the current master here) with “MACHINE=freedom-u540 bitbake demo-coreip-cli” and dd afterwards.

The system boots up fine. But if I try to restart the board, I get the following messages:

```[   86.981976] systemd-shutdown[1]: Detaching DM devices.          
[   87.423510] reboot: Restarting system                           
[  108.422889] rcu: INFO: rcu_sched self-detected stall on CPU
[  108.427707] rcu:     0-....: (5249 ticks this GP) idle=eb6/1/0x4000000000000002 softirq=3623/3623 fqs=2592
[  108.437164]  (t=5250 jiffies g=2501 q=41)                       
[  108.441158] Task dump for CPU 0:                           
[  108.444369] systemd-shutdow R  running task        0     1      0 0x00000008                                
[  108.451400] Call Trace:                                 
[  108.453849] [<ffffffe00003d968>] walk_stackframe+0x0/0xa6
[  108.459220] [<ffffffe00003dace>] show_stack+0x2a/0x34
[  108.464262] [<ffffffe00005c64a>] sched_show_task+0xba/0x10c
[  108.469813] [<ffffffe000060bb0>] dump_cpu_task+0x3e/0x48
[  108.475112] [<ffffffe000089668>] rcu_dump_cpu_stacks+0x7c/0xb4
[  108.480925] [<ffffffe000088bb8>] rcu_sched_clock_irq+0x3de/0x532
[  108.486917] [<ffffffe00008d3d4>] update_process_times+0x22/0x4c
[  108.492821] [<ffffffe00009613e>] tick_periodic+0x42/0xd6
[  108.498115] [<ffffffe0000961f2>] tick_handle_periodic+0x20/0x5e
[  108.504028] [<ffffffe000452c54>] riscv_timer_interrupt+0x32/0x3a
[  108.510018] [<ffffffe0005772fc>] do_IRQ+0xa4/0xb8
[  108.514700] [<ffffffe00003c8a4>] ret_from_exception+0x0/0xc
[  108.520256] [<ffffffe00003d056>] machine_restart+0xc/0xe
```

The whole block (except the first two lines) is repeated every 63 seconds.

Any ideas what could be the problem or how to fix it?

Best regards
Christian

davidlt, 2020-01-30:
The kernel in this release does not incl. GPIO driver and `gpio-restart` node. IIRC the new GPIO driver was approved and should be merged for 5.6 kernel (`gpio-restart` change was not yet posted). We will have those changes for the next release.

davidlt, 2020-03-02:
Closing. This is now solved in the master branch.

closed 2020-03-02 by davidlt