sifiveinc/meta-sifive #7: Can't bring up secondary CPUs

skylar-byte, 2020-03-22:
I'm running OpenSBI and 5.6-rc6 I can't get the secondary CPUs to come online.

CPU1: failed to come online

I've compiled with GCC 8.3.0 and 9.2.0.

davidlt, 2020-03-23:
5.6-rc6 kernel is not available in meta-sifive right now.

It sounds that you updated OpenSBI beyond v0.6 version and 5.6-rc6 kernel does not yet support HSM (Heart State Management) extension is SBI v0.2. The patches for the kernel are posted, but most likely will land in 5.7 cycle.

Use OpenSBI v0.6 if you want to have SMP working.

See: https://fosdem.org/2020/schedule/event/riscv_bootflow/

Could you confirm which OpenSBI version do you use?

skylar-byte, 2020-03-23:
Yup, that was the issue. I was using the master branch. I've rebuilt it with 0.6 and it worked flawlessly. Thanks!

closed 2020-03-23 by skylar-byte