sifiveinc/meta-sifive #12: Build breaks at qemu-native-5.1.0-r0 do_configure

tamood, 2020-12-06:
Log:
ERROR: qemu-native-5.1.0-r0 do_configure: Execution of '/home/tamood/test/riscv-sifive/build/tmp-glibc/work/x86_64-linux/qemu-native/5.1.0-r0/temp/run.do_configure.272720' failed with exit code 1:

ERROR: sizeof(size_t) doesn't match GLIB_SIZEOF_SIZE_T.
       You probably need to set PKG_CONFIG_LIBDIR
       to point to the right pkg-config files for your
       build target

WARNING: exit code 1 from a shell command.

davidlt, 2020-12-11:
Could you test a master branch? Could you provide more information (host OS, build tools (default system, OE/Yocto, your own built), etc.)

tamood, 2020-12-20:
Here goes the version info.

(base) tamood@tamood-virtual-machine:~/Desktop$ uname -a
Linux tamood-virtual-machine 5.4.0-56-generic #62-Ubuntu SMP Mon Nov 23 19:20:19 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
(base) tamood@tamood-virtual-machine:~/Desktop$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.1 LTS
Release:	20.04
Codename:	focal
(base) tamood@tamood-virtual-machine:~/Desktop$ 

I follow full guide at meta-sifive, branch 2020.11. I use prebuilt tools as specified. 
Previously, I was able to build 2020.6 branch, and boot linux on NexysA7 board, but this time above error halts the build process. 

The NexysA7 branch is hosted at 

https://github.com/tamood/meta-sifive/tree/u500-generic

I need to evaluate the new bootloaders which are recently replacing fsbl.

tamood, 2020-12-28:
Hi, I was able to build the new 2020.12 release on the above-mentioned machine. THe issue is considered as closed now.

closed 2020-12-28 by tamood