sifiveinc/meta-sifive #13: freedom u540 with 5.10

pankajjoshi347, 2021-02-09:


pankajjoshi347, 2021-02-09:
Hi,
I am building  image for  u540 target i have in added my own kernel recipe for 5.10.11 kernel but i am getting following error

i am attaching my kernel recipe for 5.10.11 kernel
**************************************************

require recipes-kernel/linux/linux-mainline-common.inc

LINUX_VERSION ?= "5.10.x"
KERNEL_VERSION_SANITY_SKIP="1"

BRANCH = "linux-5.10.y"

SRCREV = "v5.10.11"

SRC_URI = " \
    git://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git;branch=${BRANCH} \
"

# SRC_URI = "${KERNEL_SRC};branch=${SRCBRANCH}"

LIC_FILES_CHKSUM = "file://COPYING;md5=6bc538ed5bd9a7fc9398086aedcd7e46"
# For freedom-u540 use out-of-tree defconfig
# The upstream defconfig might contain a large number of debug options, which
# could affect the performance.
SRC_URI_append_freedom-u540 = " file://defconfig"

unset KBUILD_DEFCONFIG_qemuriscv64
unset KBUILD_DEFCONFIG_freedom-u540
**************************************************************************

/Pankaj/riscv-sifive/build$ bitbake  linux-mainline
Loading cache: 100% |########################################################################################################################################################################| Time: 0:00:00
Loaded 3780 entries from dependency cache.
Parsing recipes: 100% |######################################################################################################################################################################| Time: 0:00:00
Parsing of 2517 .bb files complete (2516 cached, 1 parsed). 3781 targets, 180 skipped, 0 masked, 0 errors.
WARNING: preferred version 5.10.11 of linux-mainline not available (for item linux-mainline)
WARNING: versions of linux-mainline available: 5.10.x+gitAUTOINC+b97134d151 5.4.x+gitAUTOINC+d4716ee875 5.8.x+gitAUTOINC+ab435ce49b
NOTE: Resolving any missing task queue dependencies
WARNING: preferred version 5.10.11 of linux-mainline not available (for item virtual/kernel)
WARNING: versions of linux-mainline available: 5.10.x+gitAUTOINC+b97134d151 5.4.x+gitAUTOINC+d4716ee875 5.8.x+gitAUTOINC+ab435ce49b
WARNING: preferred version 5.10.11 of linux-mainline not available (for item kernel-image)
WARNING: versions of linux-mainline available: 5.10.x+gitAUTOINC+b97134d151 5.4.x+gitAUTOINC+d4716ee875 5.8.x+gitAUTOINC+ab435ce49b
WARNING: preferred version 5.10.11 of linux-mainline not available (for item kernel-base)
WARNING: versions of linux-mainline available: 5.10.x+gitAUTOINC+b97134d151 5.4.x+gitAUTOINC+d4716ee875 5.8.x+gitAUTOINC+ab435ce49b
WARNING: preferred version 5.10.11 of linux-mainline not available (for item kernel-image-image.gz)
WARNING: versions of linux-mainline available: 5.10.x+gitAUTOINC+b97134d151 5.4.x+gitAUTOINC+d4716ee875 5.8.x+gitAUTOINC+ab435ce49b

Build Configuration:
BB_VERSION           = "1.49.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "riscv64-oe-linux"
MACHINE              = "freedom-u540"
DISTRO               = "nodistro"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "riscv64"
meta                 = "HEAD:68f20ac5524a5db8a09a90df796b4862cf747d3c"
meta-oe              
meta-python          
meta-multimedia      
meta-networking      
meta-gnome           
meta-xfce            = "HEAD:759058bfb832161eb0d21f5c51a49c8361e727bc"
meta-riscv           = "HEAD:b022614bc4a2ddc744ab976d87295758710382d2"
meta-clang           = "HEAD:e2d2a029ee122b9cf25f8296c6ce610bf6bebbf6"
meta-sifive          = "HEAD:a69e4d7bc44b86f30885b88a9e050cd7340aa462"

Initialising tasks: 100% |###################################################################################################################################################################| Time: 0:00:00
Sstate summary: Wanted 7 Found 0 Missed 7 Current 62 (0% match, 89% complete)
[... 96 lines cut ...]
|   LD      vmlinux
|   SYSMAP  System.map
|   OBJCOPY arch/riscv/boot/Image
|   GZIP    arch/riscv/boot/Image.gz
|   Kernel: arch/riscv/boot/Image.gz is ready
| make[2]: *** No rule to make target 'arch/riscv/boot/dts/sifive/hifive-unleashed-a00-microsemi.dtb'.  Stop.
| /home/khan/Pankaj/riscv-sifive/build/tmp-glibc/work-shared/freedom-u540/kernel-source/Makefile:1338: recipe for target 'sifive/hifive-unleashed-a00-microsemi.dtb' failed
| make[1]: *** [sifive/hifive-unleashed-a00-microsemi.dtb] Error 2
| /home/khan/Pankaj/riscv-sifive/build/tmp-glibc/work-shared/freedom-u540/kernel-source/Makefile:185: recipe for target '__sub-make' failed
| make: *** [__sub-make] Error 2
| WARNING: exit code 1 from a shell command.
| 
ERROR: Task (/home/khan/Pankaj/riscv-sifive/meta-sifive/recipes-kernel/linux/linux-mainline_5.10.11.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 476 tasks of which 475 didn't need to be rerun and 1 failed.
NOTE: Writing buildhistory
NOTE: Writing buildhistory took: 1 seconds

Summary: 1 task failed:
  /home/khan/Pankaj/riscv-sifive/meta-sifive/recipes-kernel/linux/linux-mainline_5.10.11.bb:do_compile
Summary: There were 10 WARNING messages shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

I have removed all patches as i was getting error while applying it also i dont need them.Can someone help me how i can resolve this error ?

Thanks

davidlt, 2021-02-09:
I have edited your last comment to use code section to make it readable.

There is no `hifive-unleashed-a00-microsemi.dtb` DTB in the upstream kernel. You need to modify machine file to select a proper DTB, see: https://github.com/sifive/meta-sifive/blob/2021.01/conf/machine/freedom-u540.conf#L18

`hifive-unleashed-a00-microsemi.dtb` is added by one of the patch you have removed.

pankajjoshi347, 2021-02-09:
Thank you for editing.What is the way you suggest me to make my recipe work ? i dint got exactly what i should do with this DTB.
kindly suggest

davidlt, 2021-02-09:
See the commend in the machine config. Set:
```
RISCV_SBI_FDT ?= "hifive-unleashed-a00.dtb"
```

pankajjoshi347, 2021-02-10:
kernel got compile successfully but when building image for u540 getting following error

Pankaj/riscv-sifive/build$ MACHINE=freedom-u540 bitbake demo-coreip-cli
Loading cache: 100% |########################################################################################################################################################################| Time: 0:00:00
Loaded 3780 entries from dependency cache.
Parsing recipes: 100% |######################################################################################################################################################################| Time: 0:00:00
Parsing of 2517 .bb files complete (2516 cached, 1 parsed). 3781 targets, 180 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
WARNING: preferred version 5.10.11 of linux-mainline not available (for item virtual/kernel)
WARNING: versions of linux-mainline available: 5.10.x+gitAUTOINC+b97134d151 5.4.x+gitAUTOINC+d4716ee875 5.8.x+gitAUTOINC+ab435ce49b
WARNING: preferred version 5.10.11 of linux-mainline not available (for item kernel-modules)
WARNING: versions of linux-mainline available: 5.10.x+gitAUTOINC+b97134d151 5.4.x+gitAUTOINC+d4716ee875 5.8.x+gitAUTOINC+ab435ce49b
WARNING: preferred version 5.10.11 of linux-mainline not available (for item kernel-dev)
WARNING: versions of linux-mainline available: 5.10.x+gitAUTOINC+b97134d151 5.4.x+gitAUTOINC+d4716ee875 5.8.x+gitAUTOINC+ab435ce49b
WARNING: preferred version 5.10.11 of linux-mainline not available (for item kernel-image-image.gz)
WARNING: versions of linux-mainline available: 5.10.x+gitAUTOINC+b97134d151 5.4.x+gitAUTOINC+d4716ee875 5.8.x+gitAUTOINC+ab435ce49b
WARNING: preferred version 5.10.11 of linux-mainline not available (for item kernel-image)
WARNING: versions of linux-mainline available: 5.10.x+gitAUTOINC+b97134d151 5.4.x+gitAUTOINC+d4716ee875 5.8.x+gitAUTOINC+ab435ce49b
WARNING: preferred version 5.10.11 of linux-mainline not available (for item kernel-base)
WARNING: versions of linux-mainline available: 5.10.x+gitAUTOINC+b97134d151 5.4.x+gitAUTOINC+d4716ee875 5.8.x+gitAUTOINC+ab435ce49b
WARNING: preferred version 5.10.11 of linux-mainline not available (for item kernel-module-scsi-debug)
WARNING: versions of linux-mainline available: 5.10.x+gitAUTOINC+b97134d151 5.4.x+gitAUTOINC+d4716ee875 5.8.x+gitAUTOINC+ab435ce49b
WARNING: preferred version 5.10.11 of linux-mainline not available (for item kernel-module-unix)
WARNING: versions of linux-mainline available: 5.10.x+gitAUTOINC+b97134d151 5.4.x+gitAUTOINC+d4716ee875 5.8.x+gitAUTOINC+ab435ce49b
WARNING: preferred version 5.10.11 of linux-mainline not available (for item kernel-module-binfmt-misc)
WARNING: versions of linux-mainline available: 5.10.x+gitAUTOINC+b97134d151 5.4.x+gitAUTOINC+d4716ee875 5.8.x+gitAUTOINC+ab435ce49b
WARNING: preferred version 5.10.11 of linux-mainline not available (for item kernel-module-sch-fq-codel)
WARNING: versions of linux-mainline available: 5.10.x+gitAUTOINC+b97134d151 5.4.x+gitAUTOINC+d4716ee875 5.8.x+gitAUTOINC+ab435ce49b
WARNING: preferred version 5.10.11 of linux-mainline not available (for item kernel-module-autofs4)
WARNING: versions of linux-mainline available: 5.10.x+gitAUTOINC+b97134d151 5.4.x+gitAUTOINC+d4716ee875 5.8.x+gitAUTOINC+ab435ce49b
WARNING: preferred version 5.10.11 of linux-mainline not available (for item kernel-module-ipv6)
WARNING: versions of linux-mainline available: 5.10.x+gitAUTOINC+b97134d151 5.4.x+gitAUTOINC+d4716ee875 5.8.x+gitAUTOINC+ab435ce49b
WARNING: preferred version 5.10.11 of linux-mainline not available (for item kernel-module-sctp)
WARNING: versions of linux-mainline available: 5.10.x+gitAUTOINC+b97134d151 5.4.x+gitAUTOINC+d4716ee875 5.8.x+gitAUTOINC+ab435ce49b
WARNING: preferred version 5.10.11 of linux-mainline not available (for item kernel-module-nfsd)
WARNING: versions of linux-mainline available: 5.10.x+gitAUTOINC+b97134d151 5.4.x+gitAUTOINC+d4716ee875 5.8.x+gitAUTOINC+ab435ce49b
WARNING: preferred version 5.10.11 of linux-mainline not available (for item kernel-module-x-tables)
WARNING: versions of linux-mainline available: 5.10.x+gitAUTOINC+b97134d151 5.4.x+gitAUTOINC+d4716ee875 5.8.x+gitAUTOINC+ab435ce49b
WARNING: preferred version 5.10.11 of linux-mainline not available (for item kernel-module-ip6table-filter)
WARNING: versions of linux-mainline available: 5.10.x+gitAUTOINC+b97134d151 5.4.x+gitAUTOINC+d4716ee875 5.8.x+gitAUTOINC+ab435ce49b
WARNING: preferred version 5.10.11 of linux-mainline not available (for item kernel-module-nf-nat)
WARNING: versions of linux-mainline available: 5.10.x+gitAUTOINC+b97134d151 5.4.x+gitAUTOINC+d4716ee875 5.8.x+gitAUTOINC+ab435ce49b
WARNING: preferred version 5.10.11 of linux-mainline not available (for item kernel-module-iptable-nat)
WARNING: versions of linux-mainline available: 5.10.x+gitAUTOINC+b97134d151 5.4.x+gitAUTOINC+d4716ee875 5.8.x+gitAUTOINC+ab435ce49b
WARNING: preferred version 5.10.11 of linux-mainline not available (for item kernel-module-nf-conntrack-ipv4)
WARNING: versions of linux-mainline available: 5.10.x+gitAUTOINC+b97134d151 5.4.x+gitAUTOINC+d4716ee875 5.8.x+gitAUTOINC+ab435ce49b
WARNING: preferred version 5.10.11 of linux-mainline not available (for item kernel-module-nf-defrag-ipv4)
WARNING: versions of linux-mainline available: 5.10.x+gitAUTOINC+b97134d151 5.4.x+gitAUTOINC+d4716ee875 5.8.x+gitAUTOINC+ab435ce49b
[... cut ...]
NOTE:   do_package: 0.0% sstate reuse(0 setscene, 72 scratch)
NOTE:   do_packagedata: 0.0% sstate reuse(0 setscene, 71 scratch)
NOTE:   do_package_write_ipk: 0.0% sstate reuse(0 setscene, 40 scratch)
NOTE:   do_populate_lic: 0.0% sstate reuse(0 setscene, 485 scratch)

Summary: 1 task failed:
  /home/khan/Pankaj/riscv-sifive/openembedded-core/meta/recipes-kernel/linux/kernel-devsrc.bb:do_install
Summary: There were 63 WARNING messages shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

why this could be ??

davidlt, 2021-02-10:
Kernel internal scripts have changed and thus kernel-devsrc no more works without adjusting it. If you don't plan to build kernel modules locally you can simply remove kernel-devsrc.bbappend file.

pankajjoshi347, 2021-02-11:
If i need to have modules which are built locally what i need to change ??

davidlt, 2021-02-11:
Do not know fully right now. Build `kernel-devsrc` without modifications, try building the modules (that will tell you which files are now required for out-of-tree kernel modules build), modify `kernel-devsrc` accordingly.

pankajjoshi347, 2021-02-11:
i have simply removed kernel-devsrc.bbappend  now i am getting error , ERROR: No recipes in default available for:
 /home//Pankaj/riscv-sifive/meta-sifive/recipes-kernel/linux/kernel-devsrc.bbappend

davidlt, 2021-02-11:
You have removed a wrong file. You need to remove `kernel-devsrc.bbappend`

pankajjoshi347, 2021-02-11:
Yes i have removed kernel-devsrc.bbappend from meta-sifive/recipes-kernel/linux now its throwing 

$MACHINE=freedom-u540 bitbake demo-coreip-cli
Loading cache: 100% |########################################################################################################################################################################| Time: 0:00:00
Loaded 3780 entries from dependency cache.
Parsing recipes: 100% |######################################################################################################################################################################| Time: 0:00:00
Parsing of 2516 .bb files complete (2515 cached, 1 parsed). 3780 targets, 180 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
WARNING: preferred version 5.10.11 of linux-mainline not available (for item virtual/kernel)
WARNING: versions of linux-mainline available: 5.10.x+gitAUTOINC+b97134d151 5.4.x+gitAUTOINC+5e1942063d 5.8.x+gitAUTOINC+ab435ce49b
WARNING: preferred version 5.10.11 of linux-mainline not available (for item kernel-modules)
WARNING: versions of linux-mainline available: 5.10.x+gitAUTOINC+b97134d151 5.4.x+gitAUTOINC+5e1942063d 5.8.x+gitAUTOINC+ab435ce49b
ERROR: Nothing RPROVIDES 'kernel-devsrc' (but /home/khan/Pankaj/riscv-sifive/meta-sifive/recipes-sifive/images/demo-coreip-cli.bb RDEPENDS on or otherwise requires it)
NOTE: Runtime target 'kernel-devsrc' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['kernel-devsrc']
ERROR: Required build target 'demo-coreip-cli' has no buildable providers.
Missing or unbuildable dependency chain was: ['demo-coreip-cli', 'kernel-devsrc']

Summary: There were 4 WARNING messages shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

davidlt, 2021-02-11:
Clean up recipes and remove `kernel-devsrc` too as described in the error message.

pankajjoshi347, 2021-02-11:
removed kernel-devsrc from riscv-sifive/meta-sifive/recipes-sifive/images/demo-coreip-cli.bb  it got built but at last it thrown error with 
Build Configuration:
BB_VERSION           = "1.49.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "riscv64-oe-linux"
MACHINE              = "freedom-u540"
DISTRO               = "nodistro"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "riscv64"
meta                 = "HEAD:68f20ac5524a5db8a09a90df796b4862cf747d3c"
meta-oe              
meta-python          
meta-multimedia      
meta-networking      
meta-gnome           
meta-xfce            = "HEAD:759058bfb832161eb0d21f5c51a49c8361e727bc"
meta-riscv           = "HEAD:b022614bc4a2ddc744ab976d87295758710382d2"
meta-clang           = "HEAD:e2d2a029ee122b9cf25f8296c6ce610bf6bebbf6"
meta-sifive          = "HEAD:a69e4d7bc44b86f30885b88a9e050cd7340aa462"

Initialising tasks: 100% |###################################################################################################################################################################| Time: 0:00:03
Sstate summary: Wanted 616 Found 0 Missed 616 Current 1599 (0% match, 72% complete)
NOTE: Executing Tasks
ERROR: demo-coreip-cli-1.0-r0 do_image_wic: Execution of '/home/khan/Pankaj/riscv-sifive/build/tmp-glibc/work/freedom_u540-oe-linux/demo-coreip-cli/1.0-r0/temp/run.do_image_wic.13617' failed with exit code 1:
INFO: Creating image(s)...

ERROR: _exec_cmd: install -m 0644 -D /home/khan/Pankaj/riscv-sifive/build/tmp-glibc/deploy/images/freedom-u540/hifive-unleashed-a00-microsemi.dtb /home/khan/Pankaj/riscv-sifive/build/tmp-glibc/work/freedom_u540-oe-linux/demo-coreip-cli/1.0-r0/build-wic/tmp.wic.t63345v8/boot.3/hifive-unleashed-a00-microsemi.dtb returned '1' instead of 0
output: install: cannot stat '/home/khan/Pankaj/riscv-sifive/build/tmp-glibc/deploy/images/freedom-u540/hifive-unleashed-a00-microsemi.dtb': No such file or directory

WARNING: exit code 1 from a shell command.

ERROR: Logfile of failure stored in: /home/khan/Pankaj/riscv-sifive/build/tmp-glibc/work/freedom_u540-oe-linux/demo-coreip-cli/1.0-r0/temp/log.do_image_wic.13617
Log data follows:
| DEBUG: Executing python function set_image_size
| DEBUG: 3876532.400000 = 2981948 * 1.300000
| DEBUG: 3876532.400000 = max(3876532.400000, 65536)[3876532.400000] + 0
| DEBUG: 3876533.000000 = int(3876532.400000)
| DEBUG: 3876533 = aligned(3876533)
| DEBUG: returning 3876533
| DEBUG: Python function set_image_size finished
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['/home/khan/Pankaj/riscv-sifive/meta-sifive/recipes-bsp/opensbi/opensbi_0.8.bb:do_populate_sysroot', 'virtual:native:/home/khan/Pankaj/riscv-sifive/openembedded-core/meta/recipes-devtools/e2fsprogs/e2fsprogs_1.45.6.bb:do_populate_sysroot', '/home/khan/Pankaj/riscv-sifive/meta-sifive/recipes-kernel/linux/linux-mainline_5.10.11.bb:do_populate_sysroot', 'virtual:native:/home/khan/Pankaj/riscv-sifive/openembedded-core/meta/recipes-core/update-rc.d/update-rc.d_0.8.bb:do_populate_sysroot', '/home/khan/Pankaj/riscv-sifive/openembedded-core/meta/recipes-devtools/mklibs/mklibs-native_0.1.44.bb:do_populate_sysroot', 'virtual:native:/home/khan/Pankaj/riscv-sifive/openembedded-core/meta/recipes-extended/xz/xz_5.2.5.bb:do_populate_sysroot', '/home/khan/Pankaj/riscv-sifive/openembedded-core/meta/recipes-devtools/qemu/qemuwrapper-cross_1.0.bb:do_populate_sysroot', 'virtual:native:/home/khan/Pankaj/riscv-sifive/openembedded-core/meta/recipes-devtools/bison/bison_3.7.4.bb:do_populate_sysroot', 'virtual:native:/home/khan/Pankaj/riscv-sifive/openembedded-core/meta/recipes-extended/parted/parted_3.3.bb:do_populate_sysroot', '/home/khan/Pankaj/riscv-sifive/openembedded-core/meta/recipes-kernel/kern-tools/kern-tools-native_git.bb:do_populate_sysroot', '/home/khan/Pankaj/riscv-sifive/openembedded-core/meta/recipes-devtools/libtool/libtool-native_2.4.6.bb:do_populate_sysroot', '/home/khan/Pankaj/riscv-sifive/openembedded-core/meta/recipes-core/glibc/glibc_2.32.bb:do_populate_sysroot', '/home/khan/Pankaj/riscv-sifive/openembedded-core/meta/recipes-core/glibc/cross-localedef-native_2.32.bb:do_populate_sysroot', '/home/khan/Pankaj/riscv-sifive/openembedded-core/meta/recipes-devtools/quilt/quilt-native_0.66.bb:do_populate_sysroot', 'virtual:native:/home/khan/Pankaj/riscv-sifive/openembedded-core/meta/recipes-support/bmap-tools/bmap-tools_3.5.bb:do_populate_sysroot', 'virtual:native:/home/khan/Pankaj/riscv-sifive/openembedded-core/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', '/home/khan/Pankaj/riscv-sifive/openembedded-core/meta/recipes-kernel/kmod/kmod-native_git.bb:do_populate_sysroot', '/home/khan/Pankaj/riscv-sifive/openembedded-core/meta/recipes-core/glibc/ldconfig-native_2.12.1.bb:do_populate_sysroot', 'virtual:native:/home/khan/Pankaj/riscv-sifive/openembedded-core/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', '/home/khan/Pankaj/riscv-sifive/openembedded-core/meta/recipes-devtools/gcc/gcc-runtime_10.2.bb:do_populate_sysroot', 'virtual:native:/home/khan/Pankaj/riscv-sifive/openembedded-core/meta/recipes-devtools/mtools/mtools_4.0.26.bb:do_populate_sysroot', 'virtual:native:/home/khan/Pankaj/riscv-sifive/openembedded-core/meta/recipes-devtools/automake/automake_1.16.2.bb:do_populate_sysroot', '/home/khan/Pankaj/riscv-sifive/openembedded-core/meta/recipes-devtools/gcc/gcc-cross_10.2.bb:do_populate_sysroot', '/home/khan/Panka [... cut ...]
NOTE:   do_package: 0.0% sstate reuse(0 setscene, 15 scratch)
NOTE:   do_packagedata: 0.0% sstate reuse(0 setscene, 19 scratch)
NOTE:   do_package_write_ipk: 0.0% sstate reuse(0 setscene, 162 scratch)
NOTE:   do_populate_lic: 0.0% sstate reuse(0 setscene, 1 scratch)

Summary: 1 task failed:
  /home/khan/Pankaj/riscv-sifive/meta-sifive/recipes-sifive/images/demo-coreip-cli.bb:do_image_wic
Summary: There were 62 WARNING messages shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

davidlt, 2021-02-11:
Modify https://github.com/sifive/meta-sifive/blob/2021.01/conf/machine/freedom-u540.conf#L56 to pick a different DTB.

pankajjoshi347, 2021-02-11:
sorry forgot to update,changed with hifive-unleashed-a00.dtb . But i want to build kernel modules too locally can you help ??

paul-walmsley-sifive, 2021-02-11:
Hi Pankaj,

Please contact your SiFive sales representative to arrange for further assistance.

Thanks

closed 2021-02-11 by paul-walmsley-sifive

pankajjoshi347, 2021-02-12:
copied image with the mentioned command but image is not booting further
SiFive FSBL:       2018-03-20
HiFive-U serial #: 00000127

what could be the reason ??