sifiveinc/meta-sifive #16: Default FDT file is not set, nor is USE_PREBOOT.

xnox, 2021-03-16:
I'm trying to ensure that extlinux.conf generated on ubuntu, specifies the location for each ubuntu kernel of the devicetree files, such that for each menu entry a matching one is loaded. Thus also allowing to easily build / install / test new kernels with custom dtbs.

However doing so results in:

```
Retrieving file: /lib/firmware/5.11.0-1002-generic/device-tree/<NULL>-hifive_unmatched_fu7
Failed to load '/lib/firmware/5.11.0-1002-generic/device-tree/<NULL>-hifive_unmatched_fu7
```

And similar. I think by default u-boot tries to use $FDTDIR/$SOC-$BOARD but for unmatched at least, $SOC is not defined, and whilst $BOARD is defined, it is not what is actually in use.

shouldn't configs specify

```
CONFIG_DEFAULT_FDT_FILE="sifive/hifive-unleashed-a00.dtb"
CONFIG_DEFAULT_FDT_FILE="sifive/hifive-unmatched-a00.dtb"
```
???

Also I've noticed that unmatched does not use `CONFIG_USE_PREBOOT=y` by default. But unleashed did. Not sure if that's intentional or not. I guess with unmatched you might be thinking that external devicetree should always be used which is fair enough.

xnox, 2021-03-16:
Interesting that CONFIG_DEFAULT_DEVICE_TREE is set.

davidlt, 2021-03-16:
I like the idea of
```
CONFIG_DEFAULT_FDT_FILE="sifive/hifive-unleashed-a00.dtb"
```
We also would need to modify `CONFIG_EXTRA_ENV_SETTINGS` in the config headers to add `"fdtfile=" CONFIG_DEFAULT_FDT_FILE "\0" ` IIRC PXE code will not attempt to use `soc` and `board` to assemble possible DTB path if `fdtfile` is set in the environment or is part of the label.

IIUC `CONFIG_DEFAULT_DEVICE_TREE` is for U-Boot DTS internal file. Note that U-Boot DTB contains additional overlays that are not used by Linux, but it shouldn't conflict with Linux DTB. The DTS is synced from the Linux tree regularly back to U-Boot, but that usually means that DTB is out-of-sync from Linux and latest stuff is not incl. This is the reason why it's advised to load DTB from external place (e.g. `/boot`) especially while things are being upstreamed.

IIRC `fdtcontroladdr` is exactly that internal U-Boot DTB with U-Boot overlays. If all fails above PXE would go for `fdt_addr` (location of built-in DTB within FW). We could probably enable it back, but I would wait until all (almost all) DT bindings are approved by Linux upstream.

xnox, 2021-03-16:
@davidlt 

Just tested

```
Description: set default FDT filenames to load from boot disk
Bug: https://github.com/sifive/meta-sifive/issues/16

Index: u-boot-2021.01+dfsg/configs/sifive_fu540_defconfig
===================================================================
--- u-boot-2021.01+dfsg.orig/configs/sifive_fu540_defconfig
+++ u-boot-2021.01+dfsg/configs/sifive_fu540_defconfig
@@ -1,3 +1,4 @@
+CONFIG_DEFAULT_FDT_FILE="sifive/hifive-unleashed-a00.dtb"
 CONFIG_RISCV=y
 CONFIG_SPL_GPIO_SUPPORT=y
 CONFIG_SYS_MALLOC_F_LEN=0x3000
Index: u-boot-2021.01+dfsg/configs/sifive_hifive_unmatched_fu740_defconfig
===================================================================
--- u-boot-2021.01+dfsg.orig/configs/sifive_hifive_unmatched_fu740_defconfig
+++ u-boot-2021.01+dfsg/configs/sifive_hifive_unmatched_fu740_defconfig
@@ -1,3 +1,4 @@
+CONFIG_DEFAULT_FDT_FILE="sifive/hifive-unmatched-a00.dtb"
 CONFIG_RISCV=y
 CONFIG_SPL_GPIO_SUPPORT=y
 CONFIG_SYS_MALLOC_F_LEN=0x3000
Index: u-boot-2021.01+dfsg/include/configs/sifive-fu540.h
===================================================================
--- u-boot-2021.01+dfsg.orig/include/configs/sifive-fu540.h
+++ u-boot-2021.01+dfsg/include/configs/sifive-fu540.h
@@ -72,6 +72,7 @@
 	"type_guid_gpt_loader2=" TYPE_GUID_LOADER2 "\0" \
 	"type_guid_gpt_system=" TYPE_GUID_SYSTEM "\0" \
 	"partitions=" PARTS_DEFAULT "\0" \
+	"fdtfile=" CONFIG_DEFAULT_FDT_FILE "\0" \
 	BOOTENV \
 	BOOTENV_SF
 
Index: u-boot-2021.01+dfsg/include/configs/sifive-hifive-unmatched-fu740.h
===================================================================
--- u-boot-2021.01+dfsg.orig/include/configs/sifive-hifive-unmatched-fu740.h
+++ u-boot-2021.01+dfsg/include/configs/sifive-hifive-unmatched-fu740.h
@@ -84,6 +84,7 @@
 	"type_guid_gpt_loader2=" TYPE_GUID_LOADER2 "\0" \
 	"type_guid_gpt_system=" TYPE_GUID_SYSTEM "\0" \
 	"partitions=" PARTS_DEFAULT "\0" \
+	"fdtfile=" CONFIG_DEFAULT_FDT_FILE "\0" \
 	BOOTENV
 
 #define CONFIG_PREBOOT 
```

And it works like a charm for Ubuntu, so will upload that into our u-boot shortly. Would be nice to have that included in meta-sifive and/or upstream u-boot.

xnox, 2021-04-13:
I'm against this for now, as implementing this breaks compat with older extlinux.conf without fdtdir set, setting which in turn breaks booting with qemu.

I will seek further advice from u-boot upstream about this.

closed 2021-04-13 by xnox

davidlt, 2021-04-13:
The patch you posted above doesn't touch QEMU configs headers or defconfigs. On QEMU platforms you need to use QEMU generated DT and passed to the OpenSBI. For this reason you could use `if` statement to set `fdtfile` only if it's not QEMU machine.

mengzhuo, 2021-06-11:
How can I resolve this issue? 
@xnox 

I've `chroot /mnt`  `u-boot-update` and it's not changed the file.

```
#U_BOOT_DEFAULT="l0"
#U_BOOT_ENTRIES="all"
#U_BOOT_MENU_LABEL="Debian GNU/Linux"
#U_BOOT_PARAMETERS="ro quiet"
U_BOOT_ROOT="/dev/nvme0n1p1"
#U_BOOT_TIMEOUT="50"
U_BOOT_FDT="/lib/firmware/5.11.0-1007-generic/device-tree/sifive/hifive-unmatched-a00.dtb"
#U_BOOT_FDT_DIR="/usr/lib/linux-image-"
```

xnox, 2021-06-11:
> How can I resolve this issue?
> @xnox
>
> I've `chroot /mnt` `u-boot-update` and it's not changed the file.
>
> ```
> #U_BOOT_DEFAULT="l0"
> #U_BOOT_ENTRIES="all"
> #U_BOOT_MENU_LABEL="Debian GNU/Linux" [...]

Use u-boot build for unmatched as provided by Ubuntu for the unmatched images. See https://cdimage.ubuntu.com/releases/21.04/release/

xnox, 2021-06-11:
> The patch you posted above doesn't touch QEMU configs headers or defconfigs. On QEMU platforms you need to use QEMU generated DT and passed to the OpenSBI.

Correct. But above patch to u-boot allows to use either (u-boot qemu, u-boot unleashed, u-boot unmatched) to boot correctly of the same root filesystem and identical extlinux.conf. Such that on qemu no dtb is loaded from disk, but for unleashed and unmatched the matching dtb is loaded from the disk.

Without this patch in all the uboots, it is not possible to build a universal rootfs, that is bootable by any u-boot on any platform.