sifiveinc/meta-sifive #18: do_rootfs fail

risonwung, 2021-06-15:
I want to build qemuriscv64 image using this original source code, but encounted this problem: meta-sifive ERROR: demo-coreip-cli-1.0-r0 do_rootfs: The postinstall intercept hook 'update_gtk_immodules_cache' failed, details in /home/rison/riscv-sifive/build/tmp-glibc/work/qemuriscv64-oe-linux/demo-coreip-cli/1.0-r0/temp/log.do_rootfs.
Any one knows how to solve this?

risonwung, 2021-06-15:
I have solved this issue. There are two points need be changed. First point, add "mkdir -p $D${libdir}/gtk-3.0/3.0.0" into update_gtk_immodules_cache. Second, add "mkdir -p $D/packages" into update_mine_database.

davidlt, 2021-06-15:
I have never seen this issue before. Could you send a patch for a review to the upstream?

tprrt, 2022-07-26:
This issue is fixed.

closed 2022-07-26 by tprrt