sifiveinc/meta-sifive #21: Vanilla upstream kernel uses rev1 dtb for unmatched

xnox, 2021-07-29:
Can you please submit 

https://github.com/sifive/meta-sifive/blob/2021.06/recipes-kernel/linux/files/0002-riscv-sifive-fu740-cpu-1-2-3-4-set-compatible-to-sif.patch

https://github.com/sifive/meta-sifive/blob/2021.06/recipes-kernel/linux/files/0003-riscv-sifive-unmatched-update-for-16GB-rev3.patch

upstream?

Note this issue is being discussed on linux-riscv mailing list right now. See https://lore.kernel.org/linux-riscv/f23711df-4357-f405-1672-5ce2fee6f093@kylinos.com.cn/T/#eee2d3d9292ed8364146798242284d548ab41fd0e

Which simply tries to bump 8GB to 16GB without any other changes.

CC: @davidlt @xypron

xypron, 2021-07-29:
On 7/29/21 2:35 PM, Dimitri John Ledkov wrote:
> Can you please submit
>
> https://github.com/sifive/meta-sifive/blob/2021.06/recipes-kernel/linux/files/0002-riscv-sifive-fu740-cpu-1-2-3-4-set-compatible-to-sif.patch [...]

Where are the new compatible strings used? Is there a matching driver
change?

>
> https://github.com/sifive/meta-sifive/blob/2021.06/recipes-kernel/linux/files/0003-riscv-sifive-unmatched-update-for-16GB-rev3.patch
> <https://github.com/sifive/meta-sifive/blob/2021.06/recipes-kernel/linux/files/0003-riscv-sifive-unmatched-update-for-16GB-rev3.patch>
>
> upstream?
>
> Note this issue is being discussed on linux-riscv mailing list right
> now. See
> ***@***.***/T/#eee2d3d9292ed8364146798242284d548ab41fd0e
> ***@***.***/T/#eee2d3d9292ed8364146798242284d548ab41fd0e>
>
> Which simply tries to bump 8GB to 16GB without any other changes.

The 5.13.5 kernel runs fine without changing voltages but only shows 8
GiB. If the voltage changes are really related to increasing the RAM
size, a good explanation in the commit message is needed. Otherwise the
voltage changes should be in a separate patch.

Best regards

Heinrich

xnox, 2021-07-29:
However @ColinIanKing tells me that @VincentZWC is maybe already looking into upstreaming this.

ColinIanKing, 2021-07-29:
> However @ColinIanKing tells me that @VincentZWC is maybe already looking into upstreaming this.

Apologies, this is a misunderstanding of something I told @xnox. I was referring to another bug.

davidlt, 2021-07-29:
The regulator driver for DA9063 will not work until v5.15 + some additional changes (not in meta-sifive, DT) would be needed.

Right now meta-sifive has DA9063 (incl. sub-functions) disabled by default. If you enable DA9063 right now (full thing) regulators will fail. If you also add `vcc-supply` to eeprom and temperature sensors those will not work if regulators driver fails.

So the existing regulators changes in DT should be changed with v5.15 kernel (full current mode support in DA9063 where power outputs are merged, like on Unmatched).

I am planning to send a new DT for 3A0 and 3B0 boards. That's something I am experimenting/testing on my board.

closed 2024-02-16 by tprrt