sifiveinc/meta-sifive #23: Patch out of sync with upstream: 0002-riscv-sifive-unmatched-update-for-16GB-rev3.patch

xypron, 2021-08-19:
Part of the changes in

https://github.com/sifive/meta-sifive/blob/909e6a67b0d1a6bc64adcb54848952c207c77cb0/recipes-kernel/linux/files/0002-riscv-sifive-unmatched-update-for-16GB-rev3.patch

have been accepted as 

https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/arch/riscv/boot/dts/sifive/hifive-unmatched-a00.dts?id=d09560435cb712c9ec1e62b8a43a79b0af69fe77

upstream.

Please, synchronize the patches in the SDK with upstream patches.

nanonyme, 2021-09-11:
The problem seems to be patch was partially upstreamed. It would probably make sense to split the patch into two files so unmerged part would apply.

tprrt, 2024-02-16:
Done

closed 2024-02-16 by tprrt