sifiveinc/meta-sifive #26: Unmatched: race condition in U-Boot's PCI stack

stefanct, 2021-09-03:
I was able to get Unmatched to boot from a SATA drive attached to a SIL3132 controller in the PCIe slot. To that end, I have enabled the following config options and enabled the SATA boot target:

    diff --git a/configs/sifive_unmatched_defconfig b/configs/sifive_unmatched_defconfig
    index 4b370fe31a..e6e28ea810 100644
    --- a/configs/sifive_unmatched_defconfig
    +++ b/configs/sifive_unmatched_defconfig
    @@ -49,2 +49,9 @@ CONFIG_DM_USB=y
     CONFIG_USB_XHCI_HCD=y
     CONFIG_USB_XHCI_PCI=y
    +CONFIG_AHCI=y
    +CONFIG_AHCI_PCI=y
    +CONFIG_SATA=y
    +CONFIG_LIBATA=y
    +CONFIG_SATA_SIL=y
    +CONFIG_CMD_LSBLK=y
    +CONFIG_CMD_SATA=y
    diff --git a/include/configs/sifive-unmatched.h b/include/configs/sifive-unmatched.h
    index 4fad69bb19..51a9b67103 100644
    --- a/include/configs/sifive-unmatched.h
    +++ b/include/configs/sifive-unmatched.h
    @@ -45,6 +45,7 @@
     #ifndef CONFIG_SPL_BUILD
     #define BOOT_TARGET_DEVICES(func) \
      func(NVME, nvme, 0) \
    +	func(SATA, sata, 0) \
      func(USB, usb, 0) \
      func(MMC, mmc, 0) \
      func(PXE, pxe, na) \

However, that is not enough because no SATA device is detected. I have tested this with two (completely) different SATA SSDs on both of the controller's ports:

    Hit any key to stop autoboot:  0 
    PCIE-0: Link up (Gen1-x8, Bus0)

    Device 0: unknown device
    No SATA device found!
    starting USB...

Neither repeated executions of `pci enum` nor any of the sata- or block-related commands show anything useful. Naturally, I tried enabling verbose outputs... and surprisingly the problem went away - an instance of a Heisenbug. I played around and ended up delaying PCIe probing a bit:

    diff --git a/drivers/pci/pci-uclass.c b/drivers/pci/pci-uclass.c
    index ce2eb5da2c..5e8f151fcb 100644
    --- a/drivers/pci/pci-uclass.c
    +++ b/drivers/pci/pci-uclass.c
    @@ -1058,6 +1058,9 @@ static int pci_uclass_pre_probe(struct udevice *bus)
            bus->parent->name);
      hose = dev_get_uclass_priv(bus);
     
    +	// printf("%s, bus=%d/%s, parent=%s\n", __func__, dev_seq(bus), bus->name, bus->parent->name);
    +	mdelay(4);// Not OK: 2, 3, OK: 4, 5, 10, 100
    +
      /*
       * Set the sequence number, if device_bind() doesn't. We want control
       * of this so that numbers are allocated as devices are probed. That

This function is called for each... PCI device? I am not exactly sure but definitely multiple times, so the sum of waited time is bigger but in the range of a few 10ths ms. I have tested adding (long) delays at other places that get executed during PCI+SATA enumeration, i.e. at the beginning of `sata_probe()`, `ahci_pci_probe()`, and `pcie_sifive_probe()` to no avail so I gave up searching further.

davidlt, 2021-09-03:
I highly suggest to report issues/suggest patches to U-Boot mailing list as that's where development is happening.

There is a patch (Aug 25th) suggested to add SATA support into boot order: https://lists.denx.de/pipermail/u-boot/2021-August/458984.html

Seems to be very similar to yours. We know from previous reports that not all SATA cards (or chipsets) work with Unmatched. It could also be a compatibility issue between chipsets, but anything else.

Have you tried modifying delays in https://github.com/u-boot/u-boot/blob/master/drivers/pci/pcie_dw_sifive.c ?

Especially in `pcie_sifive_wait_for_link` giving more time to wait for link training?

Again, I would expect to get more suggestions on U-Boot development mailing list.

tprrt, 2022-07-26:
It is no longer relevant.

closed 2022-07-26 by tprrt