sifiveinc/meta-sifive #34: gdb fails to compile when lzma is available on the host claiming that the library is missing

edolnx, 2021-11-05:
My host has liblzma installed and as such the recipe for gdb in riscv-sifive/openembedded-core/meta/recipes-devtools/gdb tries to compile in lzma support even if it's not yet built for the target.  Is there some override option that should be set to fix this?

```| checking for liblzma... no
| configure: error: missing liblzma for --with-lzma
| make[1]: *** [Makefile:9702: configure-gdb] Error 1
| make[1]: Leaving directory '/home/caperry/src/riscv64/riscv-sifive/build/tmp-glibc/work/riscv64-oe-linux/gdb/10.2-r0/build-riscv64-oe-linux'
| make: *** [Makefile:854: all] Error 2
| ERROR: oe_runmake failed
| WARNING: /home/caperry/src/riscv64/riscv-sifive/build/tmp-glibc/work/riscv64-oe-linux/gdb/10.2-r0/temp/run.do_compile.3435449:187 exit 1 from 'exit 1'
| WARNING: Backtrace (BB generated script):
|       #1: bbfatal_log, /home/caperry/src/riscv64/riscv-sifive/build/tmp-glibc/work/riscv64-oe-linux/gdb/10.2-r0/temp/run.do_compile.3435449, line 187
|       #2: die, /home/caperry/src/riscv64/riscv-sifive/build/tmp-glibc/work/riscv64-oe-linux/gdb/10.2-r0/temp/run.do_compile.3435449, line 171
|       #3: oe_runmake, /home/caperry/src/riscv64/riscv-sifive/build/tmp-glibc/work/riscv64-oe-linux/gdb/10.2-r0/temp/run.do_compile.3435449, line 166
|       #4: autotools_do_compile, /home/caperry/src/riscv64/riscv-sifive/build/tmp-glibc/work/riscv64-oe-linux/gdb/10.2-r0/temp/run.do_compile.3435449, line 161
|       #5: do_compile, /home/caperry/src/riscv64/riscv-sifive/build/tmp-glibc/work/riscv64-oe-linux/gdb/10.2-r0/temp/run.do_compile.3435449, line 156
|       #6: main, /home/caperry/src/riscv64/riscv-sifive/build/tmp-glibc/work/riscv64-oe-linux/gdb/10.2-r0/temp/run.do_compile.3435449, line 200
```

davidlt, 2021-11-05:
https://github.com/openembedded/openembedded-core/blob/master/meta/recipes-devtools/gdb/gdb-common.inc#L39

The configuration file has:
```
PACKAGECONFIG:append:pn-gdb = " xz"
```
You could remove this option and that will disable lzma for now.

edolnx, 2021-11-05:
Line 39 in my file looks nothing like that, but commenting it out works while giving a minor QA warning.

jim-wilson, 2021-11-16:
I have a patch in https://github.com/sifive/freedom-u-sdk/issues/148#issuecomment-965983466 and I've submitted it upstream
https://lists.openembedded.org/g/openembedded-core/message/158320

jim-wilson, 2021-11-23:
My patch was committed upstream last week.
https://github.com/openembedded/openembedded-core/commit/56623848f45cf475f909beb75209323a89837169

tprrt, 2022-11-15:
This issue no longer occurs.

closed 2022-11-15 by tprrt