sifiveinc/meta-sifive #37: how to install apt ?

alongL, 2021-12-06:
I find there is no apt-get program. 
It's very hard to install some program easily.
Can we install apt-get program in this OS ?

jim-wilson, 2021-12-06:
OpenEmbedded is a build it yourself system.  You can build apt, but there won't be any database of packages to install unless you build that too.

You might be better off with a desktop distro.  Ubuntu has an official release for Unleashed/Unmatched.
[https://wiki.ubuntu.com/RISC-V](https://wiki.ubuntu.com/RISC-V)
That is probably the easiest one to use.  There are multiple threads discussing this at forums.sifive.com.

There are also unofficial images for Debian, Fedora, and OpenSuse but not as easy to use as the Ubuntu ones.

alongL, 2021-12-07:
Got it.
Ubuntu has some unneeded software, which will  make the OS slow.

closed 2021-12-07 by alongL