sifiveinc/meta-sifive #40: Kernel panic - not syncing: Fatal exception in interrupt

pullannagari-babu, 2022-03-07:
Hello Team,

I built an unmatched image by using the 2021.02 branches, when I was testing the tables, I'm facing a kernel panic issue. 

Please can you help me to fix this issue?
randomly it's coming kernel panic issue.

$ iptables -L -v

# [Fetched 3651 B in 0s (6615 B/s)
[ 70.566379] Unable to handle kernel paging request at virtual address 0000005f837c3064
[ 70.573535] Oops [#1]
[ 70.575783] Modules linked in: ccm rtl8192ee btcoexist rtl_pci rtlwifi mac80211 cfg80211 libarc4 fuse
[ 70.584995] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G W 5.10.41 #1
[ 70.592374] epc: ffffffdf837c3064 ra : ffffffe00025da46 sp : ffffffe001603b80
[ 70.599495] gp : ffffffe001810070 tp : ffffffe00160a380 t0 : 0000000000000040
[ 70.606705] t1 : 0000000000000000 t2 : 000000000026776f s0 : ffffffe001603c00
[ 70.613913] s1 : 0000000000000000 a0 : 000000000000003b a1 : ffffffe0872c06e0
[ 70.621123] a2 : 0000000000001000 a3 : 0000000000000000 a4 : ffffffdf837c3064
[ 70.628333] a5 : 000000000000003b a6 : ffffffe080200248 a7 : ffffffe080200270
[ 70.635541] s2 : ffffffe001603c0c s3 : 0000000000000001 s4 : 000000000000003b
[ 70.642752] s5 : 0000000000000002 s6 : ffffffe0018151e0 s7 : ffffffe087520800
[ 70.649961] s8 : ffffffe0017dbab8 s9 : ffffffe0017dba70 s10: 0000000000000000
[ 70.657170] s11: ffffffe087782100 t3 : 0000000000000002 t4 : 0000000000000402
[ 70.664378] t5 : ffffffe00168e3d0 t6 : ffffffe00168e3d8
[ 70.669677] status: 0000000200000100 badaddr: 0000005f837c3064 cause: 000000000000000c
[ 70.677610] ---[ end trace 2d84ae89b2c60abb ]---
[ 70.682193] Kernel panic - not syncing: Fatal exception in interrupt
[ 70.688527] SMP: stopping secondary CPUs
[ 70.692449] ---[ end Kernel panic - not syncing: Fatal exception in interrupt ]---

Thanks
Babu

tprrt, 2022-07-26:
This issue is fixed.

closed 2022-07-26 by tprrt

pricardo7, 2022-10-30:
Hi!
What was the workaround?

tprrt, 2022-10-31:
Hi,

This issue no longer occurs on latest releases.
So, I suggest you use a newer version.

Kind regards,
Thomas