sifiveinc/meta-sifive #43: Upstreaming chassis fan / LED patches

tianon, 2022-09-13:
Hi!  Thanks for maintaining this repository. :heart:

I'm sorry if this is already noted somewhere (I looked quite a bit, but if I missed it I would really appreciate a link!) but is there already a plan or active effort to continue upstreaming patches, especially around u-boot and the chassis fan controls?

(It seems like something like https://forums.sifive.com/t/chassis-fan-header-not-working/4789/15 is probably also needed for the kernel to be able to control the fans, but just having them on at all has been a really dramatic difference for my otherwise pure-Debian-builds system and they'd understandably prefer to see patches upstreamed than maintain them just for the distribution :blush:)

tprrt, 2024-02-16:
Done

closed 2024-02-16 by tprrt

tianon, 2024-02-16:
Thank you!!! :heart: